rfetick/MPU6050_light #1: Different gyro value at startup and after reset

Titibo26, 2020-07-24:
Hi there,
I'm using esp-wroom32 and i just noticed that when we turn the power ON (from usb or from a battery) calibration coefs are 2x higher than what they should be. measurments are also 2 times higher.
When i reset the board via the reset bouton, calibration coef are goods, 2 times less than before. Measurments are perfect.

This behavior is hapening every time.

exemple :
when i turn it ON it's : (coefX = 4 / coefY = 2 / coefZ = 2)
Each time i reset it's : (coefX = 2 / coefY = 1 / coefZ = 1)

I tried to reset the MPU, only the gyro etc... but nothing works. I'm not sure if it is related to the mpu or to the ESP32 but i never had this kind of problem with other i2c device (nor icm20948 nor mpu9050...)

Thank you,

rfetick, 2020-07-31:
Hello and thank you for reporting the issue.

Did you try to use the same MPU and another micro-controller?

Or similarly did you try with the same micro-controller but another MPU6050 library? This library is based on the [Tockn](https://github.com/Tockn/MPU6050_tockn)'s one. You could try with his library and see if the problem persists. It might help to narrow down the origin of the issue (micro-controller, MPU or library).

Kind regards,

Romain

Titibo26, 2020-08-02:
Hello, thank you for your answer.

I didn't try with an other µc. I only tried with different esp32 devboard (but all being esp32 onboard).
The same problem appears with tockn's library.
It's been a long time now, but i think i tried with others library. I'll try again next week to be sure it is not related to the lib. 

I'm feeling like it's related to the µc but i can't find why. 
Also, if i set gyroOffset manually without using the calib function it seems to works fine.

I'll do some tests next week and keep you in tuch.
Thank you,

Thibaud

rfetick, 2020-08-04:
On my side I tried the code with an Arduino Uno and Nano, maybe there is a compatibility issue with ESP32. The error I may think about in the `calcGyroOffsets( )` function is a problem of address. A factor of 2 in the values could be due to a shift of one bit in the address.

The strange part is that your issue happens on powering on, and not on resetting...

Romain

Titibo26, 2020-08-05:
I don't think the problem is directly in the _calcGyroOffsets( )_ because in fact, values wich are read in it (with : wire->requestFrom((int)MPU6050_ADDR, 6);) are twice bigger at startup than after a reset. So wrong gyro offset are more a consequence of an other problem.

It seems that calling twice "begin()" function is resolving my issue ! (but no one of the reset function resolved this).

To resume : 
mpu6050.begin(); --> won't work at startup 

BUT

mpu6050.begin();
mpu6050.begin(); --> will work at startup ! (but its uggly :-) )

rfetick, 2020-08-09:
Thanks for the trick of calling `begin` twice.
I am working on it even though I will have difficulties to find the bug since my Arduino does not reproduce the error. However I let the issue open in case someone else experience the same issue on ESP32.

audiocrush, 2020-11-08:
Hi,
I have an issue that my ESP8266 crashes several times and then is able to calibrate the gyro offset and after that I get one reading and it stays basically the same no matter how much I move it...

WiFi-Setup.WiFi-Mode: Client
IP address: 192.168.1.79
Calculating gyro offset, do not move MPU6050

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffcf0 end: 3fffffc0 offset: 01a0
3ffffe90:  60000314 00000001 3ffef4e0 402141b4  
3ffffea0:  3ffef4e0 00000008 00000000 40214253  
3ffffeb0:  3ffef4e0 00000000 00000000 3ffef2aa  
3ffffec0:  00000001 00000005 3ffef4e0 402143dc  
3ffffed0:  00000001 3ffef228 3ffef4e0 40214333  
3ffffee0:  3ffef227 00000068 401005ee 3fffff30  
3ffffef0:  fffffff4 0000039f 00000006 402145b8  
3fffff00:  fffffff4 0000039f 3ffef225 40211104  
3fffff10:  007a1243 847e415a 3ffef044 4021112f  
3fffff20:  401055b1 00075f5c 3ffef044 402112a5  
3fffff30:  c5c09d4f 45356134 c3885012 3ffe9189  
3fffff40:  3fffff3c e0000000 c0710713 40213724  
3fffff50:  402146f1 000003e8 3ffe9189 3ffe9189  
3fffff60:  00000064 3ffef044 3ffef354 4020563b  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  40216cd8 4f01a8c0 feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef49c  
3fffffa0:  3fffdad0 00000000 3ffef45c 40213834  
3fffffb0:  feefeffe feefeffe 3ffe8540 40101039  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

WiFi-Setup.WiFi-Mode: Client
IP address: 192.168.1.79
Calculating gyro offset, do not move MPU6050
Done!

TEMPERATURE  : 36.50
ACCELERO   X : -0.00	Y : -0.00	Z : -0.00
GYRO       X : -0.00	Y : -0.00	Z : -0.00
ACC ANGLE  X : -144.74	Y : 35.26
ANGLE      X : -144.74	Y : 35.26	Z : -0.01
=======================================================

TEMPERATURE  : 36.50
ACCELERO   X : -0.00	Y : -0.00	Z : -0.00
GYRO       X : -0.00	Y : -0.00	Z : -0.00
ACC ANGLE  X : -144.74	Y : 35.26
ANGLE      X : -144.74	Y : 35.26	Z : -0.01
=======================================================

TEMPERATURE  : 36.50
ACCELERO   X : -0.00	Y : -0.00	Z : -0.00
GYRO       X : -0.00	Y : -0.00	Z : -0.00
ACC ANGLE  X : -144.74	Y : 35.26
ANGLE      X : -144.74	Y : 35.26	Z : -0.01
=======================================================

TEMPERATURE  : 36.50
ACCELERO   X : -0.00	Y : -0.00	Z : -0.00
GYRO       X : -0.00	Y : -0.00	Z : -0.00
ACC ANGLE  X : -144.74	Y : 35.26
ANGLE      X : -144.74	Y : 35.26	Z : -0.01
=======================================================

rfetick, 2020-11-10:
The library seems to have issues with ESP. Did you try with the Github [jrowberg](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050) or [tockn](https://github.com/Tockn/MPU6050_tockn) libraries? Since my library is based on the Tockn's one, you might get same kind of errors, however you might be able to make it work with other libraries. Please tell me if one of these libraries work. It might help to narrow down the issue

rfetick, 2020-11-11:
Some hints are discussed in [this issue](https://github.com/tockn/MPU6050_tockn/issues/3) of the Tockn library

augustosc, 2021-01-23:
Hello Rfetik. I had the same problem using an Arduino Uno. I found that in the power up the GYRO_CONFIG register contains the value 0x0 after the MPU.begin (), indicating a 250 degree /s scale. When resetting, the GYRO_CONFIG register will have a value of 0x8 after MPU.begin (), indicating the correct scale of 500 degrees /s.
I fixed the error by programming the right scale directly in the GYRO_CONFIG register with the value 0x8, after the MPU.begin () , using the Wire commands.
It is possible that the Value 0x0 after the power up results from a possible failure in the management of the PWR_MGMT_1 register in the elaboration of the MPU.begin () method.

rfetick, 2021-01-23:
Hello augustosc,
Thanks for this comment, it is very interesting for debugging this issue. Maybe the call sequence in the `MPU6050.begin( )` function of the library is wrong. Could you open the `MPU6050_light.cpp` file of the library and switch the lines of the `MPU6050::begin` function from
```cpp
setGyroConfig(gyro_config_num);
setAccConfig(acc_config_num);
byte status = writeData(MPU6050_PWR_MGMT_1_REGISTER, 0x01); // check only the last connection with status
```
to
```cpp
byte status = writeData(MPU6050_PWR_MGMT_1_REGISTER, 0x01); // check only the last connection with status
setGyroConfig(gyro_config_num);
setAccConfig(acc_config_num);
```
Please also disable your trick of setting `GYRO_CONFIG` after the `begin( )` function and tell me if the patch is solving the issue. I am waiting forward to see if this works!

Kind regards,

Romain

augustosc, 2021-01-23:
Hi rfetick
Thank you for your prompt reply.
1-I made a simple sketch  only with initialization and readings of the MPU6050 registers. I use your mpu.readData() method, better than my Wire commands. The section that matters is below:

Wire.begin();
 //XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
 //XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
 // XXXXXXXXXXX    read registers before MPU.begin()

  BT.println("Registers before mpu.begin()");
  
 readReg = mpu.readData(pwwgmgtReg);
  BT.print("PWR_MGMT_1 --> 0x");BT.println(readReg);

  readReg = mpu.readData(configReg);
  BT.print("CONFIG --> 0x");BT.println(readReg);

  readReg = mpu.readData(smplRtReg);
  BT.print("SMPLRT_DIV --> 0x");BT.println(readReg);

  readReg = mpu.readData(gyroConfReg);
  BT.print("GYRO_CONFIG --> 0x");BT.println(readReg);

  readReg = mpu.readData(accConfReg);
  BT.print("ACCEL_CONFIG --> 0x");BT.println(readReg);

  
  
 //--------------------------------  initialize MPU6050
  BT.println("inicializa MPU6050");

 
  byte status = mpu.begin();
  BT.print(F("MPU6050 status: "));
  BT.println(status);
  while(status!=0){ } // stop everything if could not connect to MPU6050

//XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
 //XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
 // XXXXXXXXXXX    read registers after MPU.begin()

  BT.println("Registers after mpu.begin()");
  readReg = mpu.readData(pwwgmgtReg);
  BT.print("PWR_MGMT_1 --> 0x");BT.println(readReg);

  readReg = mpu.readData(configReg);
  BT.print("CONFIG --> 0x");BT.println(readReg);

  readReg = mpu.readData(smplRtReg);
  BT.print("SMPLRT_DIV --> 0x");BT.println(readReg);

  readReg = mpu.readData(gyroConfReg);
  BT.print("GYRO_CONFIG --> 0x");BT.println(readReg);

  readReg = mpu.readData(accConfReg);
  BT.print("ACCEL_CONFIG --> 0x");BT.println(readReg);
 
  BT.println(F("Calculating offsets, do not move MPU6050"));
  mpu.calcOffsets(); // gyro and accelero
 
  heading=leAngulo();
  BT.print("Heading: ");BT.println(heading);

2-I did and registered tests without changes in the library. Watch,
in the log,  in the power-up, the values ​​of the mpu registers after mpu.begin ():
PWR_MGMT_1 -> 0x1 and GYRO_CONFIG -> 0x0, the latter value causing the error.

Watch, after the reset, the values ​​of the registers after mpu.begin ():
PWR_MGMT_1 -> 0x1 and GYRO_CONFIG -> 0x8, the correct value.

[... 119 lines cut ...]
Heading: -0.04

PWR_MGMT_1 --> 0x1
CONFIG --> 0x0
SMPLRT_DIV --> 0x0
GYRO_CONFIG --> 0x8
ACCEL_CONFIG --> 0x0
inicializa MPU6050
MPU6050 status: 0
Registers after mpu.begin()
PWR_MGMT_1 --> 0x1
CONFIG --> 0x0
SMPLRT_DIV --> 0x0
GYRO_CONFIG --> 0x8
ACCEL_CONFIG --> 0x0
Calculating offsets, do not move MPU6050
Heading: -2.53

5-These are the results of the test you requested.
These were not exhaustive tests, but they may indicate a possible solution for the error.
Thank you for your good work on this library MPU6050_light, 
which greatly simplifies the use of the MPU6050.

Sincerely
Augusto Sherman

rfetick, 2021-01-24:
Thanks for these great tests. I will implement you point number 4 and check it on my own hardware. You can also open a pull request on github with your modifications, if you prefer.

Kind regards,

Romain

augustosc, 2021-01-24:
Hi Romain
thank you to try point number 4. Please, give me feedback. I hope the changes work well.
Regards
Augusto.

augustosc, 2021-01-24:
Hi Romain,
Thank you to try point 4. Please, give me feedback.  I think it is not necessary the pull a  request. I hope the changes work well.

Best regards
Augusto

rfetick, 2021-01-24:
Hello,

I implemented your changes and it worked also on my Arduino! I committed the changes on the library on github (it is available for download). However it is not available yet on the Arduino library manager, I will push it soon for a new release.

I will let the issue open for some time to see if it helps for anyone, especially with ESP32 microcontroller.

Thanks again for your help,

Romain

augustosc, 2021-01-24:
My pleasure to be able to help.

Thank you
Augusto