rfetick/MPU6050_light #3: Accelerometer Z Axis offset calibration

audiocrush, 2020-11-11:
Hi,

I have three MPU-6050 boards and they all have a different offset on the Z-Axis. like 0.3 Gs, one is 0.1 Gs and the third is 0.4Gs off which is bit annoying xD Subsequently all the values like ACC Angle and Angle are wrong.
I have worked around this by passing a float to the update function to compensate for that, but I think that it would be kind of cool to be able to just calibrate the Z-Axis offset during initialization?
Is that possible on the MPU-6050?

rfetick, 2020-11-11:
Thanks for the comment. This can be done by implementing an accelerometer calibration function (as for the gyroscopes). The values given by the MPU6050 are measured during a short time, and then averaged to find the offset. I will work on this!

audiocrush, 2020-11-14:
Hi,
yaay I just saw your update and tried it. Works like a charm.
Thank you a lot!
This improves the ease of use greatly :)
As well as the accuracy of the angles of course.
Have a great weekend :)

rfetick, 2020-11-14:
Perfect! I will create soon a new minor release including this small improvement on accelero calibration

closed 2020-11-14 by rfetick