rfetick/MPU6050_light #4: Glitch on X angle value when flipping from 180 to -180

Mael19, 2020-12-25:
Hello, I have an issue when the MPU6050 is upside down and the angle passes from 180 to -180 and vice versa. I get some "parasite values" (see code) that I'd like to remove, does anyone know how to do that?
Cheers and merry Christmas!

```
X : 174.68	Y : 0.70	Z : -13.56
X : 174.85	Y : 0.68	Z : -13.56
X : 175.05	Y : 0.69	Z : -13.56
X : 175.24	Y : 0.69	Z : -13.56
X : 175.38	Y : 0.67	Z : -13.56
X : 175.46	Y : 0.68	Z : -13.56
X : 175.62	Y : 0.68	Z : -13.56
X : 175.81	Y : 0.68	Z : -13.56
X : 175.91	Y : 0.68	Z : -13.56
X : 175.99	Y : 0.70	Z : -13.55
X : 176.12	Y : 0.70	Z : -13.54
X : 176.31	Y : 0.69	Z : -13.54
X : 176.55	Y : 0.69	Z : -13.53
X : 176.75	Y : 0.70	Z : -13.53
X : 176.92	Y : 0.68	Z : -13.52
X : 177.09	Y : 0.69	Z : -13.52
X : 177.12	Y : 0.70	Z : -13.52
X : 177.13	Y : 0.69	Z : -13.53
X : 177.17	Y : 0.70	Z : -13.53
X : 177.26	Y : 0.69	Z : -13.53
X : 177.35	Y : 0.68	Z : -13.53
X : 177.43	Y : 0.67	Z : -13.53
X : 177.54	Y : 0.66	Z : -13.53
X : 177.69	Y : 0.66	Z : -13.53
X : 177.85	Y : 0.67	Z : -13.53
X : 178.03	Y : 0.67	Z : -13.54
X : 178.26	Y : 0.67	Z : -13.54
X : 178.58	Y : 0.63	Z : -13.55
X : 171.81	Y : 0.61	Z : -13.57
X : 126.24	Y : 0.59	Z : -13.59        // parasite X angle from here, should go straight to approx. -180
X : 92.74	Y : 0.41	Z : -13.59
X : 57.65	Y : 0.58	Z : -13.61
X : 26.60	Y : 0.71	Z : -13.63
X : 20.41	Y : 0.75	Z : -13.63
X : 1.22	Y : 0.76	Z : -13.63
X : -22.61	Y : 0.78	Z : -13.63
X : -43.28	Y : 0.75	Z : -13.63
X : -61.26	Y : 0.74	Z : -13.63
X : -76.89	Y : 0.74	Z : -13.63
X : -90.43	Y : 0.75	Z : -13.63
X : -100.61	Y : 0.74	Z : -13.63
X : -104.25	Y : 0.73	Z : -13.63
X : -114.19	Y : 0.74	Z : -13.63
X : -122.81	Y : 0.75	Z : -13.63
X : -130.30	Y : 0.74	Z : -13.63
X : -136.77	Y : 0.72	Z : -13.63
X : -142.41	Y : 0.74	Z : -13.63
X : -147.29	Y : 0.74	Z : -13.63
X : -151.54	Y : 0.74	Z : -13.63
X : -155.25	Y : 0.74	Z : -13.63
X : -158.04	Y : 0.74	Z : -13.63
X : -160.86	Y : 0.75	Z : -13.63
X : -163.33	Y : 0.76	Z : -13.63
X : -165.46	Y : 0.76	Z : -13.63
X : -167.33	Y : 0.76	Z : -13.63
X : -168.96	Y : 0.77	Z : -13.63
X : -170.37	Y : 0.78	Z : -13.63
X : -171.59	Y : 0.77	Z : -13.63
X : -172.66	Y : 0.79	Z : -13.63
X : -173.57	Y : 0.78	Z : -13.63
X : -174.24	Y : 0.78	Z : -13.63
X : -174.95	Y : 0.81	Z : -13.63
X : -175.57	Y : 0.79	Z : -13.63
X : -176.08	Y : 0.78	Z : -13.63
X : -176.54	Y : 0.77	Z : -13.63
X : -170.59	Y : 0.78	Z : -13.63
X : -171.77	Y : 0.77	Z : -13.63
X : -172.80	Y : 0.75	Z : -13.63
X : -173.69	Y : 0.75	Z : -13.63
X : -174.47	Y : 0.76	Z : -13.63
X : -175.04	Y : 0.75	Z : -13.63
X : -175.63	Y : 0.75	Z : -13.63
X : -176.15	Y : 0.76	Z : -13.63
X : -176.60	Y : 0.76	Z : -13.63
X : -176.97	Y : 0.76	Z : -13.63
X : -177.31	Y : 0.75	Z : -13.63
X : -177.60	Y : 0.76	Z : -13.63
X : -177.85	Y : 0.77	Z : -13.63
X : -171.58	Y : 0.75	Z : -13.63
X : -172.63	Y : 0.75	Z : -13.63
X : -173.43	Y : 0.75	Z : -13.63
X : -174.26	Y : 0.74	Z : -13.63
X : -174.96	Y : 0.74	Z : -13.63
X : -175.57	Y : 0.75	Z : -13.63
X : -176.09	Y : 0.77	Z : -13.63
X : -176.54	Y : 0.77	Z : -13.63
X : -176.93	Y : 0.79	Z : -13.63
X : -177.29	Y : 0.79	Z : -13.63
X : -177.58	Y : 0.77	Z : -13.63
```

rfetick, 2020-12-27:
Hello,

With the definition of the angles, there is an angular jump when the MPU6050 is turned upside-down. However the complementary filter integrates the gyroscope data, that provides a quite smooth angular estimation. In this point the jump is then smoothed by the complementary filter. It is possible to give less weight to the gyroscope data, but then the measure relies more on the accelerometer data, that is noisy...

I don't know how to avoid this angular jump without using quaternions (that is not the formalism used in this library...)

I will think about this, any idea is also welcome.

Kind regards,

Romain

rfetick, 2021-01-09:
The issue has been solved in the new version of the code available for download on Github. Do not update the code from the Arduino library manager (in case you use it) since I did not push yet the new version to the library manager.

You should now have a sharp transition from +180° to -180° on the X angle.

The credit for the solution goes to [Edgar Bonet's answer to this issue](https://github.com/gabriel-milan/TinyMPU6050/issues/6).

Kind regards,

Romain

Mael19, 2021-01-10:
Hello, thank you so much for your fix! I just tried the new version and it seems to work perfectly! 

Best regards,
Mael

closed 2021-01-10 by Mael19