rfetick/MPU6050_light #6: Unexpected X, Y value range when run upside down

daarashaw, 2021-01-08:
Hi,

I have an MPU6050 breakout board mounted on a Teensy 3.2.
The GetAngle example program runs fine, but when starting the program with the MPU6050 upside down the X and Y values only have a range of -25 to 25 (when rotated by 90° in either direction). Z values are unaffected.

Is there a calculation I should apply to get expected value ranges, or would this be a library specific issue?

rfetick, 2021-01-09:
Hello Daara-S,

The calibration has to be performed when the MPU6050 is in upside-up position (and not upside-down)!!! If you do so with your MPU upside-down, the accelerometer reads approximately +1g on the Z axis, whereas it should be -1g (multiplication by -1 of the values, whereas the calibration supposes only an offset).

**What you can do**: Save the calibration whereas the MPU6050 is upside-up, and use these calibration values when it is started upside-down. The library will state that your MPU is upside-down, but with correct angular ranges. Tell me if this solves the issue.

**What I can do**: Add a upside-down mode that reverts the Z axis. So your MPU6050 can work upside-down as if it was upside-up.

Kind regards,

Romain

rfetick, 2021-01-09:
I just added a new feature to manage upside-down mounting. Please download the new library from github. Don't update the library from the Arduino library manager (in case you use it) since I didn't push the new version yet on the Arduino library manager.

Then have a look at the the new [getAngle.ino](https://github.com/rfetick/MPU6050_light/blob/master/examples/GetAngle/GetAngle.ino) example. There is a line to uncomment to manage upside-down mounting. The MPU6050 will behave as if it was mounted upside-up.

Please tell me if this solves your issue.

Romain

Daara-S, 2021-01-09:
Hi Romain,

Thank you for the quick response, as well as the quick fix. The new feature works perfectly, I'm getting the same expected response as when the sensor is the right way up.

Best wishes,
Daara

closed 2021-01-09 by Daara-S