rfetick/MPU6050_light #7: Absolute orientation

b0x4it, 2021-01-23:
I am getting zero angles at startup. So, regardless of the initial orientation of the MPU6050, the angles are zero. Moving it updates the angles, but they are relative to the initial orientation. How can I change it to show the absolute angles with respect to gravity?

Thanks you

rfetick, 2021-01-24:
Hello b0x4it,
You should calibrate once the MPU6050 when it is laying on a flat surface, and retrieve the offsets with the appropriate functions as `getAccXoffset` (you can print them). Afterwards, write a new script that doesn't compute calibrations, but forces the offset values with `setAccOffsets` for example.

b0x4it, 2021-01-25:
Hello rfetick,
Do you have any sample code? 
I am not experienced with MPU. 
Thank you

rfetick, 2021-01-25:
Hello,
Maybe you can have a look at the [documentation](https://github.com/rfetick/MPU6050_light/blob/master/documentation_MPU6050_light.pdf) of the library, paying particular attention to sections 4 and 5.

You will have to write two scripts, using two different `setup` functions:
* Offset printing script when the MPU6050 is calibrated on a flat surface (to be run once): the setup contains the `calcOffset` function. Then retrieve the offsets with the `getAccXoffset` and `getGyroXoffset` (similar on Y and Z). Finally print these offsets and write them down.
* Offset setting script when the MPU6050 is started in any position: use the `setAccOffsets` and `setGyroOffsets` functions in the setup instead of `calcOffset` to force the offset values. Provide the value previously computed.

Kind regards,

Romain

b0x4it, 2021-01-26:
Thank you for the documentation.

Do you know why the below code can get the absolute angles without any calibration and dealing with offsets?

`//Written by Ahmet Burkay KIRNIK
//TR_CapaFenLisesi
//Measure Angle with a MPU-6050(GY-521)

#include<Wire.h>

const int MPU_addr = 0x68;
int16_t AcX, AcY, AcZ, Tmp, GyX, GyY, GyZ;

int minVal = 265;
int maxVal = 402;

void mGyro_setup() {
  Wire.begin();
  Wire.beginTransmission(MPU_addr);
  Wire.write(0x6B);
  Wire.write(0);
  Wire.endTransmission(true);
}
void mGyro_loop() {
  Wire.beginTransmission(MPU_addr);
  Wire.write(0x3B);
  Wire.endTransmission(false);
  Wire.requestFrom(MPU_addr, 14, true);
  AcX = Wire.read() << 8 | Wire.read();
  AcY = Wire.read() << 8 | Wire.read();
  AcZ = Wire.read() << 8 | Wire.read();
  int xAng = map(AcX, minVal, maxVal, -90, 90);
  int yAng = map(AcY, minVal, maxVal, -90, 90);
  int zAng = map(AcZ, minVal, maxVal, -90, 90);

  AngleX = RAD_TO_DEG * (atan2(-yAng, -zAng) + PI);
  AngleY = RAD_TO_DEG * (atan2(-xAng, -zAng) + PI);
  //z = RAD_TO_DEG * (atan2(-yAng, -xAng) + PI);
}`

rfetick, 2021-01-26:
I apologize but I don't debug here other people code. There might be specialized websites to do so, maybe give a try on [https://arduino.stackexchange.com/](https://arduino.stackexchange.com/)

However for your information, the code you show appears quite clumsy and I wouldn't use it. It simply ignores calibrations, and the mapping seems extremely dirty...

If you want to ignore calibrations, you can just comment the `mpu.calcOffsets` line from my [example script](https://github.com/rfetick/MPU6050_light/blob/master/examples/GetAngle/GetAngle.ino) and it will do the job.

Kind regards,

Romain

b0x4it, 2021-01-27:
Here is what I did:

- edited `setup ` to include the following:

```
mpu.calcOffsets(); // gyro and accelero

accXoffset = mpu.getAccXoffset();
accYoffset = mpu.getAccYoffset();
accZoffset = mpu.getAccZoffset();
gyroXoffset = mpu.getGyroXoffset();
gyroYoffset = mpu.getGyroYoffset();
gyroZoffset = mpu.getGyroZoffset();

Serial.println(accXoffset, 10);
Serial.println(accYoffset, 10);
Serial.println(accZoffset, 10);

Serial.println(gyroXoffset, 10);
Serial.println(gyroYoffset, 10);
Serial.println(gyroZoffset, 10);
```

- upload the code and got the offset values
- then modified `setup` to use the offsets as follow:

```
mpu.setAccOffsets(0.0609169938, 0.0123232426, -0.0538652353);
mpu.setGyroOffsets(0.1213435084, 0.1675420254, -2.7240309715);
```
It works! Thank you. 

One issue that I noticed is that, when the absolute angles go beyond 90 degree then start decreasing to zero. the same is happening on the negative side. How can I adjust the calculation to get 0-180 degree or even better 0-360 degree angles?

rfetick, 2021-01-27:
The library has been written to follow the [Euler angles convention](https://en.wikipedia.org/wiki/Euler_angles#Signs,_ranges_and_conventions). It is not possible to modify this.

b0x4it, 2021-01-27:
a question regarding calibration: when we calibrate it and leave the MPU 6050 on a flat surface, does the flat surface has to be exactly zero degree in real world? What if the flat surface is slightly tilted? Does this affect the calibration?

rfetick, 2021-01-28:
Yes it should be zero degree in real world during calibration.

closed 2021-01-28 by rfetick