rfetick/MPU6050_light #8: Where can i find and change the pin connection

LedHat, 2021-01-27:
Hey,thanks for the library but where can i find to wich pins you connected the mpu and where can i possibly change it?

rfetick, 2021-01-27:
The MPU6050 should be connected to the default SDA and SCL pins of the micro-controller. Maybe you can have a look at the Arduino [Wire library](https://www.arduino.cc/en/Reference/Wire) to know the corresponding pin numbers.

Kind regards,

Romain

LedHat, 2021-01-28:
I see, how stupid the question sounds afterward. I was just confused because of the other I2c pins for an magnetometer. Is it planned to make magnetometer possible in this library too?

rfetick, 2021-01-28:
Unfortunately I don't plan to include magnetometer. The main reason is that I don't have access to a magnetometer, so it will be impossible to debug for me. The second reason is that I try to keep the library efficient but as simple as possible, it is often a compromise between simplicity and interesting new features.

Kind regards,

Romain

LedHat, 2021-01-29:
Alright, no problem

closed 2021-01-30 by rfetick