rfetick/MPU6050_light #12: How to switch the readable register?

Soupman123, 2021-04-26:
Hello, how would I change whether this library is reading from register 0x68 to 0x69. If this is able to be done it could allow multiple sensors to be attached to the same program.

rfetick, 2021-04-27:
Hello Soupman123,

So far the MPU address is defined in the header of the library without possibility to modify it for the user (hard coding). You may change the following line in the header file if you only use one MPU6050:
`#define MPU6050_ADDR 0x68`

The other option would be to modify slightly the constructor of MPU6050 to allow the user to provide its own address. I will put this in my TODO list!

Best regards,

Romain

rfetick, 2021-05-30:
Address can now be modified by user. See notes on [release version 1.2.1](https://github.com/rfetick/MPU6050_light/releases/tag/v1.2.1)

closed 2021-05-30 by rfetick