rfetick/MPU6050_light #14: Zeroing sensor values

ivanbo97, 2021-04-28:
Hello!

First I would like to thank you for your great contribution and your effort in simplyfing the whole process of sensor's data retrieval. I was wondering if there is a way to safely zero the values of angleX, angleY, angleZ after MPU6050 has already been calibrated in setup()? Can we just add setter methods to class MPU6050 ?

Best regards,
Ivan

rfetick, 2021-04-28:
Hello,

You cannot set to zero `angleX`, `angleY` and `angleZ` since they are estimations from the raw data (accelerometer and gyro). If the calibration is done correctly these values should be close to zero just after start. They can suffer from raw data measurement noise, but there is nothing to do to solve this.

Kind regards,
Romain

ivanbo97, 2021-04-28:
Actually I was wondering if I can do second calibration. My initial purpose is to make my robot's front to point North which requires some additional steps because MPU6050 doesn't have magnetomer. They are as follow:
1. Retrieve current gps coordinates.
2. Move the robot x meters forward.
3. Once again retrieve gps coordinates.
4. Based on the two coordinates I use additional formula to calculate the current rotation around the z-axis (azimuth in degrees)
   For example the result is 255° and the MPU6050 initially shows 0°.
5. I want the robot to face north (360°), so I call a function written by me: rotate(360-255);
   After function execusion the robot is facing north but MPU6050 will show rotation around z :  ≈0° + (360°-255°) ≈ 105°.
   
Is it possible to calibrate MPU6050 once again and zero its values (especially rotation around z-axis), because thus the MPU6050 output value for rotaton around z will correspond to compass values.

rfetick, 2021-04-28:
OK I understand better your problem now. Your GPS data give you the absolute heading angle (based on the recipe you explained), whereas MPU6050 only provides a relative one. Both are related by an offset, so you can compute the offset and apply it to the `angleZ` value given by the library.
```cpp
// compute offset only once
float heading_absolute = getHeadingFromGPS();
float heading_offset = heading_absolute - mpu.getAngleZ();
// now apply offset on all your measurements
float unbiased_heading = mpu.getAngleZ() + heading_offset;
```

You are then implicitly performing data fusion between GPS and MPU6050. Different devices provide measurements or estimates of the state of your robot, and you want to efficiently merge these data to get the "best" estimate of the state.

I might revise my opinion, but I don't think I would implement this directly inside the MPU6050 library, because it looks like data fusion with other components, that is out of the scope of the library. Its scope is only to provide the best estimate given internal data. Then the user can obviously decide to merge data as he/she likes in his/her Arduino script. I hope you understand my point.

Kind regards,

Romain

ivanbo97, 2021-04-29:
Quite helpful answer! Thank you for your quick response.

closed 2021-04-29 by ivanbo97