rfetick/MPU6050_light #19: Two MPUs

Maco-coder, 2021-07-06:
Hi there!

This isn't really an issue. I was wondering how I could make the code work for two MPU sensors connected to the same Arduino. I haven't found a way to set an address for each of them. At this moment, the code compiles and uploads, but my 2 MPUs are displaying very close (if not the same) values. Any help would be appreciated!! Thanks.

_#include <MPU6050_light.h>
#include "Wire.h"

MPU6050 mpu6050(Wire);
MPU6050 mpu6050B(Wire);

unsigned long timer = 0;

void setup() {
  
  Serial.begin(9600);
  
  Wire.begin()      ;
  mpu6050.begin()   ;
  mpu6050.calcOffsets();

  Wire.begin()      ;
  mpu6050B.begin() ;
  mpu6050B.calcOffsets();
}

void loop() {
 
 mpu6050.update();
 mpu6050B.update();
 
 if((millis()-timer)>2000){ // print values every 2s
  
  Serial.print("Accelerometer 1 ");
  Serial.print("X : ");
  Serial.print(mpu6050.getAngleX());
  Serial.print("\tY : ");
  Serial.print(mpu6050.getAngleY());
  Serial.print("\tZ : ");
  Serial.print(mpu6050.getAngleZ());
  
  Serial.print("\tAccelerometer 2 ");
  Serial.print("X : ");
  Serial.print(mpu6050B.getAngleX());
  Serial.print("\tY : ");
  Serial.print(mpu6050B.getAngleY());
  Serial.print("\tZ : ");
  Serial.println(mpu6050B.getAngleZ());
  
  //digitalWrite(LEDfun, HIGH);
  timer = millis();
 }
  
}_

rfetick, 2021-07-06:
Hello,

Now the library has `setAdress` and `getAdress` methods. You can provide addresses as `0x68` and `0x69`. You can also look online on how to connect two MPU6050 to an Arduino board. Even though I never did this, it seems you need to send a logical `True` to the pin of one MPU and a `False` to the other MPU. It allows to identify which one gets which address. However I don't know more about this.

Kind regards,

Romain

Maco-coder, 2021-07-08:
Thanks a lot Romain!

The setAddress method helped me solve the issue I was having. Now I am able to measure the tilt angles of both MPUs, and this time the readings are completely different, as I move around one and I only rotate a bit (e.g., in Z) the other one.

rfetick, 2021-07-09:
Perfect, enjoy!

closed 2021-07-09 by rfetick

Gugarauj07, 2022-07-16:
Hey, Im trying to use the setAddress() Method but im still only having the data from the 0x68 device. Can somebody help me?