rfetick/MPU6050_light #20: support of 2 sensors

76nic, 2021-08-10:
Dear Romain,
I really like the work you have done on the library MPU6050_light. Would you consider to add support of 2 MPU sensors. Basically calling all the functions you have currently implemented with "mpu0x68.xxx" and "mpu0x69.xxx".
Many thanks in advance for thinking about it!
Best regards,
76nic

rfetick, 2021-08-28:
Hello 76nic,

As far I understand your issue as already been solved. Please have a look at the [previous issue here](https://github.com/rfetick/MPU6050_light/issues/19).
You can also have a look at the [release version 1.2.1](https://github.com/rfetick/MPU6050_light/releases/tag/v1.2.1) which  solved the two sensors issue.

Please tell me if this helps

Cheers,

Romain

76nic, 2021-09-08:
Dear Romain, many thanks for your responds and pointing me to the mpu.setAddress() functionality. I have a question regarding the mpu.calcGyroOffsets() function using 2 sensors in parallel. Is the calibration done for both sensors and stored if I call that function? Basically like the following:
MPU6050 mpuA(Wire);
MPU6050 mpuB(Wire);
void setup()
    Wire.begin();
    mpuA.setAddress(104);
    mpuA.begin();
    Serial.println(F("Calculating gyro offset, do not move MPU6050"));
    delay(1000);
    mpuA.calcGyroOffsets();
    Serial.println("Done.");
    
    mpuB.begin();
    Serial.println(F("Calculating gyro offset, do not move MPU6050"));
    delay(1000);
    mpuB.setAddress(105);
    mpuB.calcGyroOffsets();
    Serial.println("Done.");
void loop()
    mpuA.update();
    currentAngleX_A = mpuA.getAngleX();
    currentAngleY_A = mpuA.getAngleY();
    currentAngleZ_A = mpuA.getAngleZ();

    mpuB.update();
    currentAngleX_B = mpuB.getAngleX();
    currentAngleY_B = mpuB.getAngleY();
    currentAngleZ_B = mpuB.getAngleZ();

Many thanks,
- 76nic

76nic, 2021-09-08:
sorry - the code format turned out weird by uploading - it is meant as continues code.

eiffelblithe, 2022-03-23:
> "Dear Romain, many thanks for your responds and pointing me to the `mpu.setAddress()` functionality. I have a question regarding the `mpu.calcGyroOffsets()` function using 2 sensors in parallel. Is the calibration done for both sensors and stored if I call that function? Basically like the following:
> 
> ```c++
> MPU6050 mpuA(Wire);
> MPU6050 mpuB(Wire);
> void setup()
>     Wire.begin();
>     mpuA.setAddress(104);
>     mpuA.begin();
>     Serial.println(F("Calculating gyro offset, do not move MPU6050"));
>     delay(1000);
>     mpuA.calcGyroOffsets();
>     Serial.println("Done.");
>     
>     mpuB.begin();
>     Serial.println(F("Calculating gyro offset, do not move MPU6050"));
>     delay(1000);
>     mpuB.setAddress(105);
>     mpuB.calcGyroOffsets();
>     Serial.println("Done.");
> void loop()
>     mpuA.update();
>     currentAngleX_A = mpuA.getAngleX();
>     currentAngleY_A = mpuA.getAngleY();
>     currentAngleZ_A = mpuA.getAngleZ();
> 
>     mpuB.update();
>     currentAngleX_B = mpuB.getAngleX();
>     currentAngleY_B = mpuB.getAngleY();
>     currentAngleZ_B = mpuB.getAngleZ();
> ```
> 
> Many thanks,
> 
> * 76nic"

Hello thank you for this code but may I know how would the code be like if it is for 5 sensors and not only 2 sensors? For this code 2 registers were used but what if it is for 5 sensors? Thank you so much.

Mo-Alsehli, 2023-12-03:
To what pins should the other MPU connected to?