rfetick/MPU6050_light #24: GetAngle works fine alone but "hangs" up when embedded in other sketch

drbrodie, 2021-12-17:
Using the GetAngle code I have no problems but when the same code is embedded in a much larger sketch which calls mpu.update about every 12 ms the program freezes after a minute +-.  Does mpu.update need to be called more frequently? Can you suggest anything else which might be happening?

rfetick, 2022-01-04:
Dear drbrodie,

Are the angular values diverging? It could be a memory overflow.
Clean calibration might be needed to avoid gyroscopic drift, but I am not completely sure this is your issue.

Kind regards,

Romain

drbrodie, 2022-01-05:
Romain,
Thanks for your reply. It is not that the angles are diverging, it is that
the MPU6050 freezes up, halting the program into which it is embedded.  I
changed to the Adafruit library but the problem continues to occur although
not as often. This seems to be happening to others as well.  I am changing
to the BNO055 instead of the MPU6050 with which I have had no such problems
in the past. Thanks for your effort.