rfetick/MPU6050_light #25: AngleZ value

Mudkip909, 2022-01-03:


rfetick, 2022-01-04:
Could you please provide more details on the issue?
Why the angle should be constant? In which configuration are you running your MPU6050 device?

Mudkip909, 2022-01-04:
> Could you please provide more details on the issue?
> Why the angle should be constant? In which configuration are you running your MPU6050 device?

i have my MPU mounted in my project facing pins up. On the y axis, it shows the real angle the mpu is at, but on the z it is rather angular acceleration. I checked the code and it appears at the bottom, the z angle is not wrapped and it says “to do”.

rfetick, 2022-01-04:
Hello,

This behaviour is strange, since the values of the gyro are integrated they should give the angle and not an angular acceleration.

In fact the last line comment is `// not wrapped (to do???)` since I didn't know if it was a good idea to wrap this angle. I think it is better to not wrap it since unwrapped values allow to give an absolute angle measurement, or in other words the number of turns around the Z axis. Wrapping this value will get rid of this information.