rfetick/MPU6050_light #26: calcOffsets not working if being called outside of setup?

motedon, 2022-01-05:
Hello, I'm currently making a RC car and my remote uses the MPU6050 + this library.
I have 2 modes of control which involves the MPU: AngleX(for turning left and right) and angleY(for going forward and backwards) for the first mode and AngleZ, acting as a steering for the second mode. I know that when I turn on the remote and wait for the offset calculation, all values should be 0(assuming remote was put in the desired position). But I wanted to add a functionality to the remote where no matter the holding position, if I press a button, I can recalculate the offsets and I can start controlling the car with the new offsets. I tried doing that using the calcOffsets(true, true) method, but it doesn't work. I also tried using the setGyroOffsets(0,0,0) and setAccOffsets(0,0,0) but it also didn't work for me. what works is resetting the board(arduino mega). I could be wrong in assuming that the offset calculation in this library just sets the fields to 0 and start doing angle measurements from there, that's how I think it works, if not, a clarification from the devs will be great.
Great library and documentation, thank you for making it!

rfetick, 2022-01-05:
Hello motedon,

Thank you for this clear description of the issue. Indeed calibration should be able to be performed outside of the `setup` function. I will check this on my side.

However the library has been designed to work close to the flat regime, since it assumes gravity along the Z axis for calibration. When calibrating in any position, the gravity is not on the Z axis and the angular computations fail.

I think that you should calibrate the MPU6050 in a flat position at the beginning. Then, when the user push a button, you save the current angles given by the library. Then you remove these angles as an offset, see the code below

```cpp
float angXoffset = 0;
float angYoffset = 0;

void setup(){
  ... // initialize everything
  mpu.calcOffsets(); // calibrate MPU6050 when flat
}

void loop(){
  mpu.update();
  float angX = mpu.getAngleX() - angXoffset;
  float angY = mpu.getAngleY() - angYoffset;
  ... // do whatever you want with the angles

  if(button_pushed){
    angXoffset = mpu.getAngleX();
    angYoffset = mpu.getAngleY();
  }
}
``` 

Remember that the library works in the small angle regime, don't approach too much the 90° on the X or Y axis.

Cheers,

Romain

motedon, 2022-01-05:
Thank you very much and currently I'm using from -30 to +30 degrees, will probably won't go past 50, so I'm fine. I will try to implement your suggestion and share the results.

motedon, 2022-01-07:
Thank you very much Romain, your solution works as expected.

closed 2022-01-07 by motedon