rfetick/MPU6050_light #28: Yaw angles are constantly drifting or increasing/decreasing rapidly

Jawad-RoboLearn, 2022-02-17:
Hello, I am using your library and sending IMU data over ROSSerial to my ground station. 

Everything is going ok but yaw angle are constantly increasing/decreasing their values. Sometimes there is a drift. For e.g., if I just put the IMU at some horizontal surface, yaw angle try to increase/decrease. 

B) Do you have some calibration offsets function in your library so I will add the value of offset for my MPU6050? As I have already calculated the offsets.

motedon, 2022-02-18:
I've had that problem too, have you tried resetting the board?

rfetick, 2022-02-20:
Hello,

You might have a look at the [documentation](https://github.com/rfetick/MPU6050_light/blob/master/documentation_MPU6050_light.pdf) of the library, especially the section describing all the available methods. To answer your question, yes there is a method to define offsets, and it is highly recommanded to define offsets to avoid drifts. The yaw angle is obviously the less constrained since gravity cannot correct for drift. It is not an issue with the library, but something that cannot be solved unless you have access to other data to constrain this angle. A poor MPU6050 device or a wrong calibration will lead to strong drift on this angle.

Cheers,

Romain