rfetick/MPU6050_light #29: Pitch value changes with change in GyroZ

frozenmafia, 2022-03-04:
Thanks for your amazing library.

I am using your library for balancing drone. 
For eg my drone is at zero pitch and zero roll. But as soon as I get angular rotation in z axis. pitch and roll value increases for a moment and return backs to normal. But that moment is enough for my drone to crash. 
I would be higly obliged if you provide me any guidance to solve this issue
Thanking you.

rfetick, 2022-03-04:
Hello,

Could it be due to centrifugal force? It could be the case if the MPU6050 is far from the rotation center and at high Z-axis rotation speed. Values of gyroscopes and accelerometers at this critical moment will give you a hint on the cause. If it is indeed the centrifugal force you might see accelerometer values to increase on X or Y axis.

As a reminder, the library provides correct angular values in the case where the gravity is the major acceleration. If other forces (such as centrifugal force) become a non negligible source of acceleration, then angular computations are affected.

Another cause might be a slight misalignement of your MPU6050 with the frame of the drone. The Z-axis rotation is projected on the X and Y axis (as the sine of the angular misalignement). In this case, these are the gyroscope values on X and Y that increase at the critical moment you described.

Cheers,

Romain

cemyasam2, 2023-09-08:
> 

MPU6050_light.cpp dosyasında alttaki satırı bulun

  accX = ((float)rawData[0]) / ACC_LSB_2_G;
  accY = ((float)rawData[1]) / ACC_LSB_2_G;
  accZ = ((float)rawData[2]) / ACC_LSB_2_G;
  temp = (rawData[3] + TEMP_LSB_OFFSET) / TEMP_LSB_2_DEGREE;
  gyroX = ((float)rawData[4]) / GYRO_LSB_2_DEGSEC - gyroXoffset;
  gyroY = ((float)rawData[5]) / GYRO_LSB_2_DEGSEC - gyroYoffset;
  gyroZ = ((float)rawData[6]) / GYRO_LSB_2_DEGSEC - gyroZoffset;

yukarıdaki satırın altına  alttaki 2 satırı ekleyin böylecee gelen parazitik sıçramaları önleyebilirsiniz.
if(gyroZ>0 && gyroZ<0.4) {gyroZ=0;}
if(gyroZ<0 && gyroZ>-0.4) {gyroZ=0;}
 z için yaptıgım bu düzeltmeyi x ve y içinde uygulayın