rfetick/MPU6050_light #34: Angle Z gyro drifting

Gabriel110302, 2022-10-07:
I made the MPU6050 calibration several times and tried to change some values from the <MPU6050_light.h> library, but I cannot figure out to get the angle Z gyroscope not to drift away. Sometimes it drifts up sometimes down. I read that Earth moves 15 degree per hour so that's why it's drifts like this. What should I do to counter this?

rfetick, 2022-12-03:
Hello Gabriel110302,
The MPU6050 is a cheap device that is subject to important noise and drift. I assume that you calibrate well your device on a flat surface as required by the library. Then the error might be due to the limited sensitivity of the sensor itself. If this issue is important for you, you may need a better sensor or couple the one you have with a magnetometer to get an absolute reference around the Z angle.
Cheers,
Romain

cemyasam2, 2023-09-08:
accX = ((float)rawData[0]) / ACC_LSB_2_G;
  accY = ((float)rawData[1]) / ACC_LSB_2_G;
  accZ = ((float)rawData[2]) / ACC_LSB_2_G;
  temp = (rawData[3] + TEMP_LSB_OFFSET) / TEMP_LSB_2_DEGREE;
  gyroX = ((float)rawData[4]) / GYRO_LSB_2_DEGSEC - gyroXoffset;
  gyroY = ((float)rawData[5]) / GYRO_LSB_2_DEGSEC - gyroYoffset;
  gyroZ = ((float)rawData[6]) / GYRO_LSB_2_DEGSEC - gyroZoffset;
yukarıdaki satırın altına  alttaki 2 satırı ekleyin böylecee geln parazitik sıçramaları sönümler.
if(gyroZ>0 && gyroZ<0.4) {gyroZ=0;}
if(gyroZ<0 && gyroZ>-0.4) {gyroZ=0;}

cemyasam2, 2023-09-08:
> Merhaba Gabriel110302, MPU6050 önemli gürültü ve kaymalara maruz kalan ucuz bir cihazdır. Cihazınızı kütüphanenin gerektirdiği şekilde düz bir yüzey üzerinde iyi bir şekilde kalibre ettiğinizi varsayıyorum. O zaman hata, sensörün sınırlı hassasiyetinden kaynaklanıyor olabilir. Bu konu sizin için önemliyse, Z açısı etrafında mutlak bir referans elde etmek için daha iyi bir sensöre ihtiyacınız olabilir veya sahip olduğunuz sensörü bir manyetometreyle eşleştirebilirsiniz. Şerefe, Romain

MPU6050_light.cpp dosyasında alttaki satırı bulun

  accX = ((float)rawData[0]) / ACC_LSB_2_G;
  accY = ((float)rawData[1]) / ACC_LSB_2_G;
  accZ = ((float)rawData[2]) / ACC_LSB_2_G;
  temp = (rawData[3] + TEMP_LSB_OFFSET) / TEMP_LSB_2_DEGREE;
  gyroX = ((float)rawData[4]) / GYRO_LSB_2_DEGSEC - gyroXoffset;
  gyroY = ((float)rawData[5]) / GYRO_LSB_2_DEGSEC - gyroYoffset;
  gyroZ = ((float)rawData[6]) / GYRO_LSB_2_DEGSEC - gyroZoffset;

yukarıdaki satırın altına  alttaki 2 satırı ekleyin böylecee gelen parazitik sıçramaları önleyebilirsiniz.
if(gyroZ>0 && gyroZ<0.4) {gyroZ=0;}
if(gyroZ<0 && gyroZ>-0.4) {gyroZ=0;}